ericstoneking/42 #141: Orbit propagation consistency with SGP4

DarioTurchi, 2024-09-18:
Hello,
I'm struggling to get the simulator orbit propagation matching (to a reasonable extent) that provided by SGP4.
I'm using the following TLE for defining the SC orbit in 42 (FIXED propagation, J2 correction disabled) 
```
1 00005U 58002B   23152.00000000  .00000023  00000-0  28098-4 0  4753
2 00005  34.2682 348.7242 1859667 331.7664 019.3264 10.82419157413667
```
I'm checking the inertial position from 42 against a snippet of python code that uses `sgp4` package for propagation and `SkyCoord` for TEME->GCRS transformation.
If I start the simulation at a date close to the TLE epoch the result are kind of similar:

> TIME 2023-152-00:00:01.000000000
> ...
> SC[0].AC.GPS[0].PosN = 7.026203250431e+06 -1.387729685791e+06 8.792283880122e+03

while the python code returns
> 7016954.62905, -1430434.97952, -11210.81077

for an error of ~48 Km

However, the errors grows quickly 1 day later:
>TIME 2023-153-00:00:01.000000000
> ...
>SC[0].AC.GPS[0].PosN = -1.184078743920e+06 -6.211977654629e+06 -4.308523213335e+06

against 
> -973517.06211, -6263777.47516, -4290667,18939

for an error of ~218 Km

And the error keeps growing. One more day later
>TIME 2023-154-00:00:01.000000000
> ...
> SC[0].AC.GPS[0].PosN = -8.915224431832e+06 -1.659169376395e+06 -2.296359712620e+06 

against
> -8664364.83854, -1874539.60882, -2989955.1858

For an error of ~767 Km

Please note the coordinates are in meters and the PosN is taken from the GPS measurement for convenience, but I've taken care to set the position error for the GPS sensor to zero.

Is this kind of drift between 42-FIXED and SGP4 propagation expected ? Am I missing something in setting up the simulation configuration files ?

Thank you very much for any suggestion

ericstoneking, 2024-09-18:
Hi Dario,

You *might* have some luck by enabling Secular Orbit Drift Due to J2 in your Orb_ file, but it still may not meet your expectations.  There are many orbit propagators.  I can't match them all, so I don't try.  Your setup in 42 yields a purely Keplerian orbit, which is only an approximation for low Earth orbits.

If you want 42 to match some external orbit propagator, there is really only one way.  Use your external orbit propagator to generate a text file with time, position, and velocity.  (See 42ephem.c:SplineToPosVel for format.)  Then in your Orb_ file, choose the FILE and SPLINE options, and supply the File Name of your propagator-generated file.  Note that this defines your Reference Orbit, so you can still model perturbations from that (eg. due to aerodynamic forces or thruster forces) by setting Orbit Prop to ENCKE or COWELL in your SC_ file. 

Regards,
-Eric

DarioTurchi, 2024-09-18:
With J2 correction enabled I get a weird effect: the position at the beginning of the simulation is the same regardless of the simulation time:

TIME 2023-152-00:00:01.000000000
...
SC[0].AC.GPS[0].PosN = 7.033475140848e+06 -1.389164730448e+06 8.805214590743e+03

>TIME 2023-153-00:00:01.000000000
>...
>SC[0].AC.GPS[0].PosN = 7.033475140848e+06 -1.389164730448e+06 8.805214590772e+03

>TIME 2023-153-12:00:01.000000000
>...
>SC[0].AC.GPS[0].PosN = 7.033475140848e+06 -1.389164730448e+06 8.805214590765e+03

>TIME 2023-154-09:00:01.000000000
>...
>SC[0].AC.GPS[0].PosN = 7.033475140848e+06 -1.389164730448e+06 8.805214590766e+03

All of these are obtained from the same TLE just changing the simulation starting time ... am I missing something ?

ericstoneking, 2024-09-19:
Hi Dario,

That was a bug in the orbit initialization from TLE.  Thanks for finding it, and sorry for it being there.  See if the code I just committed behaves better.

Regards,
-Eric

DarioTurchi, 2024-09-19:
Hi Eric,
happy this has helped sort out an issue. By enabling the J2 correction after the fix, the FIXED propagation is now quite closer to SGP4 one.

 Thank you very much for the quick fix and all the effort you put into the simulator development and maintenance.

closed 2024-09-19 by DarioTurchi