ericstoneking/42 #142: 42 Cam stars over earth with black squared bacground

asela-jaops, 2024-09-18:
Sorry if this is a configuration mistake, but I am using 42 with Nos3 and I am experiencing some weird behaviour that I could not find in other shared images. It seems as if the stars are always overlapping the earth with a black square on the background as seen in the picture. 
[image]
I deployed Nos3 (and hence 42) in a clean installation of Ubuntu 22.04 with 3D acceleration enabled and 128 MB of video memory. 
Let me know if there is something else I should provide for troubleshooting. Thanks!
Thanks a lot!

ericstoneking, 2024-09-18:
Hi,

My guess is that your Ubuntu installation comes with an old version of OpenGL and/or glut.

The black boxes around the stars should be transparent, and the Earth should be drawn over the stars.  The stars showing through suggests that some depth/opacity formats are wrong.  That is what points to your install being old.  Not conclusively, but it's a good first thing to try.

If it helps, uncomment the call to CheckOpenGLProperties() about 14 lines from the end of 42glut.c.  There may be clues.

Regards,
-Eric

asela-jaops, 2024-09-18:
Thank you very much for the fast answer Eric. I will try that and keep posting here of the result. 
Thanks a lot!
Alex

asela-jaops, 2024-09-26:
[image]
After updating OpenGL to the latest version I get same issue. Miky way is over the earth. I also uncommented the cal as suggested, where should I search for the clues? Note that 42 launch is done automatically by Nos3, 
Thanks!
[image]

ericstoneking, 2024-09-26:
Hi Alex,

Disclaimers: I'm more of a dynamicist than a graphics guru, and I know next to nothing about NOS3 (other than it's cool).  That said, let's see what we can puzzle out.

That call to CheckOpenGLProperties() should dump some text to your terminal.  It should tell you things about what version of OpenGL is being used, and what extensions are supported.  I'll paste my output below as an example.

The fact that your terminal has words like "VirtualBox", "Mesa", and "Compatibility" makes me suspect that you have some middlemen between 42 and your graphics hardware.  42 uses shaders, which are little programs installed (by 42 at runtime) onto the graphics hardware, enabling some cool graphics techniques that are otherwise hard to pull off.  It may be that my shaders aren't getting along with your middlemen.  It might be possible for me to revert to a shader-free option for 42, but that would take some effort.  I'd rather encourage you to explore how Mesa and VirtualBox might be conflicting with 42.  I would hope that it's some easy-to-fix configuration option, but it may take some sleuthing.  I can do some web searching, but you should as well, since only you can experiment with your installation.

Regards,
-Eric

## Begin output of CheckOpenGLProperties
OpenGL Vendor: Apple
OpenGL Renderer: Apple M1 Pro
OpenGL Version: 2.1 Metal - 88.1
OpenGL Extensions: GL_ARB_color_buffer_float GL_ARB_depth_buffer_float GL_ARB_depth_clamp GL_ARB_depth_texture GL_ARB_draw_buffers GL_ARB_draw_elements_base_vertex GL_ARB_draw_instanced GL_ARB_fragment_program 
[...ten minutes later...] GL_EXT_texture_filter_anisotropic GL_EXT_texture_integer GL_EXT_texture_lod_bias 
GL_GLEXT_PROTOTYPES not defined!
GL_VERSION_1_3 defined
GL_ARB_multitexture defined
OpenGL Error 0: no error
OpenGL Max Texture Units: 8
## End output of CheckOpenGLProperties

haisamido, 2025-05-13:
@asela-jaops I think the 42 that's instantiated by nos3 comes from here https://github.com/nasa-itc/42 which is a fork from https://github.com/nasa-itc/42/compare/master...ericstoneking%3A42%3Amaster so perhaps it's best to ask the nos3 folks for assistance?

asela-jaops, 2025-05-14:
Thanks a lot @haisamido and @ericstoneking given the fact that no other people complained about this and that it only happened on a dedicated virtual machine, I stopped troubleshooting further. A clean installation on a separated partition worked properly on my side. If I encounter a similar issue again I will raise the bug again a spend more hours on the troubleshooting. Thanks a lot!

closed 2025-05-14 by asela-jaops