ericstoneking/42 #143: 42 configuration for multispacecrafts

girardja, 2024-09-19:
Considering 2 satellites on 2 different VM (VM1, VM2)  and another VM (VM42) for 42

Can anybody indicate us the configuration to put on each VM to have 42 operationnal ?

We have identified these four files that seems to define this configuration
Inp_IPC.txt Rx
Inp_IPC.txt Tx
Inp_Sim.txt  in InOut directory
Inp_NOS3.txt ?
Are there other configuration files ?

Are there any examples or documentation to help us.

Thank you for your help

Jack & Simon

ericstoneking, 2024-09-19:
Hi Jack and Simon,

If I read your intent correctly, you should study the Standalone scenario.  Start with 42/Standalone/Readme.txt.

Regards,
-Eric

Simon9050, 2024-09-19:
Hi Eric, 

Thank you for your quick response, 

I will try to rephrase the question as I suppose that an example can really help in this case :

We are using 42 with NOS3, we would like to adapt the 42 configuration files in the /InOut folder in order to have 2 FSW (on separate VMs) discussing with 42.

For example we have modified the Inp_Sim.txt as follow :
[image]

And we were trying to understand how we should modify the Inp_IPC.txt ... 

For example : 
[image]

1- Is this the right way to do it ? Are we missing something ? (we will keep studying the doc I promise 😉... But for now, it is not fully clear...)

2 - Concerning the Inp_IPC.txt files in the Tx, Rx folders, do we have to modify also these files or not ?  How should we specify to 42 to send the data to the 2 VMs and to listen from the 2 VMs ? 

Thank you for your help, 

Best Regards, 

Simon