ericstoneking/42 #144: GpsModel(struct SCType *S) always says Valid

sdunlap-afit, 2024-09-30:
In 42sensors.c  `GpsModel(struct SCType *S)`, `GPS->Valid` is set true inside the SampleCounter if-statement but is only set to false if the RefOrb is not Earth. Is this the intended functionality (always valid after the first update) or a bug? Thanks.

```
for(Ig=0;Ig<S->Ngps;Ig++) {
            GPS = &S->GPS[Ig];
            
            GPS->SampleCounter++;
            if (GPS->SampleCounter >= GPS->MaxCounter) {
               GPS->SampleCounter = 0;
               
               GPS->Valid = TRUE;
               ....
```

sdunlap-afit, 2024-10-01:
Thanks for getting back to me. I was going to use the Valid flag to determine if a new measurement was taken (due to slower sample times), but looking at the other sensor types (e.g., `AcGyroType`), I realize I'll need a different approach. I could use `SampleCounter == 0` in the higher level types (e.g., `GyroType`) since that seems pretty consistent. How do you normally handle sample rates? I have a private fork of 42 that I can modify (I wanted to call my C++ code directly from `42fsw.c` to keep sim times down).

Thanks,
Steve

ericstoneking, 2024-10-01:
Hi Steve,

Sounds like you want a flag or a time tag in the sensor model, that you can check and clear in your sensor processing.  For example, 

In GpsModel, when the measurement is taken, set GPS->NewMeas = TRUE.  Then, in your sensor processing...

`if (GPS->NewMeas) {
      GPS->NewMeas = FALSE;
      [...process measurement...]
}`

There are variations on the theme, depending on needs and taste.

Regards,
-Eric (he/him)

sdunlap-afit, 2024-10-01:
Thanks Eric. That's effectively what I'm doing. I'm implementing some custom error models, so I think I'll handle the flag at the same time. That solves my problem. Thank you for your time!
Steve