ericstoneking/42 #151: FSW related command I/F with external application

gardinwin, 2025-01-10:
User can control FSW-related commands using Inp_Cmd.txt.
I would like to integrate these commands with an external application.
What methods are available for this?

# FSW-related commands
%lf SC[%ld] FswTag = %s
   # %s is PASSIVE_FSW, PROTOTYPE_FSW, etc.
%lf SC[%ld] qrn = [%lf %lf %lf %lf]
%lf SC[%ld] qrl = [%lf %lf %lf %lf]
%lf SC[%ld] Cmd Angles = [%lf %lf %lf] deg, Seq = %ld wrt %c Frame
   # %c is either N or L
%lf SC[%ld].G[%ld] Cmd Angles = [%lf %lf %lf] deg
#  In the following, the (first) %s is either "Primary" or "Secondary"
%lf Point SC[%ld].B[%ld] %s Vector [%lf %lf %lf] at RA = %lf deg, Dec = %lf deg
%lf Point SC[%ld].B[%ld] %s Vector [%lf %lf %lf] at World[%ld] Lng = %lf deg, Lat = %lf deg, Alt = %lf km
%lf Point SC[%ld].B[%ld] %s Vector [%lf %lf %lf] at World[%ld]
%lf Point SC[%ld].B[%ld] %s Vector [%lf %lf %lf] at GroundStation[%ld]
%lf Point SC[%ld].B[%ld] %s Vector [%lf %lf %lf] at %s
   # Last %s is SUN, MOON, any planet, VELOCITY, or MAGFIELD
%lf Point SC[%ld].B[%ld] %s Vector [%lf %lf %lf] at SC[%ld]
%lf Point SC[%ld].B[%ld] %s Vector [%lf %lf %lf] at SC[%ld].B[%ld] point [%lf %lf %lf]
%lf Align SC[%ld].B[%ld] %s Vector [%lf %lf %lf] with %c-frame Vector [%lf %lf %lf]
   # %c-frame can be H, N, or L
%lf Align SC[%ld].B[%ld] %s Vector [%lf %lf %lf] with SC[%ld].B[%ld] vector [%lf %lf %lf]
Event Eclipse Entry SC[%ld] qrl = [%lf %lf %lf %lf]
Event Eclipse Exit SC[%ld] qrl = [%lf %lf %lf %lf]
Event Eclipse Entry SC[%ld] Cmd Angles = [%lf %lf %lf] deg, Seq = %ld wrt %c Frame
   # %c is either N or L
Event Eclipse Exit SC[%ld] Cmd Angles = [%lf %lf %lf] deg, Seq = %ld wrt %c Frame
   # %c is either N or L
%lf Set SC[%ld] RampCoastGlide wc = %lf Hz, amax = %lf, vmax = %lf
%lf Spin SC[%ld] about Primary Vector at %lf deg/sec

ericstoneking, 2025-01-10:
Inp_Cmd.txt is a text file.  Can your external application edit text files?

gardinwin, 2025-01-10:
I want to controls with external apps like cosmos
Is it possible ?

closed 2026-02-25 by ericstoneking