ericstoneking/42 #152: Planets do not render correctly, they are mostly pitch black

TheNeenk, 2025-01-16:
When I run 42 all the planets have some rendering issues, Earth included. Only a little fraction is rendered, the part directly illuminated by the Sun. 
I am running it through Windows (msys). What may the problem be and how can I solve it?

Thank you very much! 

[image]
[image]

ericstoneking, 2025-01-16:
My first guess would be that something in your system isn't getting along with OpenGL.  Are your video drivers up to date?  Did you obtain your freeglut  and glew libraries from the recommended sources?

There's a commented-out function call in 42glut.c (or 42glfw.c, if you're using GLFW, but I'm guessing you're not): CheckOpenGLProperties().  Uncomment that call, make and run.  You should get some diagnostic text in your terminal window.  Look in particular for:

GL_VERSION_1_3 defined
GL_VERSION_2_0 defined
GL_VERSION_2_1 defined
GL_ARB_multitexture defined
OpenGL Error 0: no error
OpenGL Max Texture Units: 8

For context, it looks like some of planet textures aren't rendering correctly.  These textures are stored as ppm files in the World folder, loaded into graphics memory at runtime, and rendered using multitextures and shaders.  You should be able to inspect the ppm files with a graphics app like GIMP.  Check EarthCol??.ppm, where ?? = PX, PY, PZ, MX, MY, MZ.  From your screenshot, it looks like only PX is rendering correctly.

If your textures are okay, then something is up with your system's handling of textures or shaders.  Hopefully the ideas above will get you on the right path.  Good luck.

Regards,
-Eric (he/him)

TheNeenk, 2025-01-16:
Thank you for the fast response! 

I tried different versions of glew and freegult but to no avail. I downloaded them from the sources written in the Install msys on windows txt. (The freeglut I got is from "Martin Payne's Windows binaries (MSVC and MinGW)").
The drivers are up to date. I uncommented the CheckOpenGLProperties() and got the following:

OpenGL Vendor: ATI Technologies Inc.
OpenGL Renderer: AMD Radeon PRO W7500
OpenGL Version: 4.6.0 Compatibility Profile Context 24.Q4.241029
GL_GLEXT_PROTOTYPES defined
GL_VERSION_1_3 defined
GL_ARB_multitexture defined
OpenGL Error 0: nessun errore
OpenGL Max Texture Units: 8

In the meanwhile I have installed Ubuntu on Windows and launched it from there. In this case I have no graphical issues and everything is rendered correctly. Only problem there is that it seems that I cannot select anything from the Unit Sphere View. Apart from that it is working just fine.

I will continue looking into the problem, if I'll find a solution I will post it here.