ericstoneking/42 #176: implicit declaration of function 'nanosleep'

waterbug, 2025-07-27:
Compilation error:

gcc -std=c11 -g -O0 -fpic -Wall -Wshadow -Wno-deprecated -Wno-unused-variable -Wno-unused-but-set-variable -Wno-stringop-overread  -I /usr/include/GL/  -I ./Include/ -I ./Kit/Include/ -I ./Kit/Source/   -D _ENABLE_GUI_ -D _USE_GLUT_ -D _USE_SHADERS_      -c ./Source/42exec.c -o ./Object/42exec.o
./Source/42exec.c: In function 'AdvanceTime':
./Source/42exec.c:117:13: error: implicit declaration of function 'nanosleep' [-Wimplicit-function-declaration]
  117 |             nanosleep(&SleepDuration,NULL);
      |             ^~~~~~~~~
make: *** [Makefile:259: Object/42exec.o] Error 1

waterbug, 2025-07-28:
Putting
"#define _POSIX_C_SOURCE 199309L"
at the top of Source/42exec.c fixed it.
(... result of asking Claude about it ... he had other ideas, too, but that was the simplest fix.)

ericstoneking, 2025-07-28:
Okay, let me throw this one to the collective wisdom.  IIRC, I replaced "microsleep" with "nanosleep" and added "-std=c11" to fix deprecation warnings that someone's compiler was throwing.  It is always my intent to make 42 be as portable as reasonably possible, so I seriously don't want to trade one compiler version issue for another.  This "199309L" looks very specific to me.  Is there a "-std=foo" that y'all can agree on so that I can quit outguessing compilers and go back to doing dynamics?

Thanks,
-Eric